StanfordVL/iGibson #193: iGibson PYTHON PATH instruction in unclear in Installation page

vbalaji21, 2022-04-04:
Hi,

I am doing the iGibson and ROS Integration https://stanfordvl.github.io/iGibson/ros_integration.html.
The third step regarding PYTHON PATH is unclear.
I have attached the screenshot below.

I did not understand this - 

For PYTHONPATH: /usr/lib/python2.7/dist-packages/, /opt/ros/kinetic/lib/python2.7/dist-packages(ROS python libraries), <anaconda installation root>/envs/py2-igibson/lib/python2.7/site-packages(iGibson dependencies) and <iGibson root> need to be in PYTHONPATH in this exact order.

roberto-martinmartin, 2022-04-14:
Hi,
we are going to revamp the iG-ROS bridge this weekend and try to update it to ROS Noetic. That should make things easier.

Wrt. your question:
1) there is no screenshot
2) the line you copied is saying that in your environmental variable PYTHONPATH (you can see it by executing `echo $PYTHONPATH`) the four listed paths should appear in the specified order. e.g., if the iGibson root path appears the first, things won't work well. 

I think the instructions are clear, but in any case, the current code may be outdated, even if the PYTHONPATH is correct.

We will let you know on Monday if the new version of the bridge (and updated instructions) are ready.

Thanks for your patience!

fxia22, 2022-04-18:
Hi @vbalaji21 , I am still in the process of investigating ROS neodic + igibson 2.1, I will let you know in a few days. Thank you for your patience.

-Fei

vbalaji21, 2022-04-19:
@roberto-martinmartin and @fxia22 Thanks a lot for the update. 
Regarding the Python Path question. Thanks a lot for your answer, We solved it.

We tried the ROS neotic and igibson 2.1 integration. We see that the system launches but the robot does not move properly to the velocity commands sent by move_base as explained in the email. We tried to debug it in multiple ways but we are still at the same point.

We are looking forward for your investigation and the new version of the bridge. 
Thanks a lot for the update.

-Vignesh

fxia22, 2022-04-23:
Yes, we observed similar issue as you do regarding the robot does not move properly to the velocity commands sent by move_base. 

We are working on a solution. @cgokmen do you mind taking a look at why the differential drive interface amplify the angular velocity? Does the differential drive interface accept angular velocity command in rad/s?

fxia22, 2022-04-23:
@cgokmen see this video for example:

https://user-images.githubusercontent.com/5158896/164944487-7d16bcea-38fa-4e95-93de-62bd103ae954.mp4

vbalaji21, 2022-05-05:
Thank you @roberto-martinmartin, @fxia22, @cgokmen  we fixed the problem. The problem was in the file igibson/robots/turtlebot.py where the convention for left and right wheel joints in the low level controller was reversed. Hence, we changed it similar back to normal convention. Hence we changed the line 36 of the file from return np.array([1, 0]) to return np.array([0, 1]). I have done it in my cloned public repository https://github.com/vbalaji21/iGibson/commit/f289b9323dd2a70338279ee3d87c6e0595a20792#diff-6329b481dbbef19b16938973d65c4ed536b33c4542474a64dbab07d068feda19. We can contribute this patch to your repository and may be it will be useful for others who wanted to use this. Let us know :)

vbalaji21, 2022-05-05:
Further, We fixed 2 more problems -
1) Fetch importing problem (Dimension mismatch)
2) Fetch robot arm tucking
 
We will be happy to contribute this also as a patch to your repository so that it will be helpful for all the others in the community. Let us know :)

JasonLiu324, 2022-07-28:
Hi, Im tring to run the igibson-ros demo but I found that the CMakeList.txt is empty to some extend. Because when I run catkin_make, it seems that it doesn't compile anything and the igibson-ros package also couldn't be located by roslaunch. Could you please offer me some instructions on this?