StanfordVL/iGibson #194: ERROR:libEGL

WJJLBJ, 2022-04-07:
i got such problems:

libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
Unable to initialize EGL
Command '['/home/server/iGibson/gibson2/core/render/mesh_renderer/build/test_device', '4']' returned non-zero exit status 1.

and i believe my nvidia-driver is good

can you help me fix the problem?
thank you!

mjlbach, 2022-04-13:
What's the version of the nvidia driver? Try downgrading, see: https://github.com/StanfordVL/iGibson/issues/189#issuecomment-1096080326

ChengshuLi, 2022-06-10:
Feel free to re-open the issue if you have further questions.

closed 2022-06-10 by ChengshuLi