StanfordVL/iGibson #196: Does 3D rendering mode use real units?

CharlesAverill, 2022-04-18:
According to the [Renderer docs](https://stanfordvl.github.io/iGibson/renderer.html), the 3D rendering mode provides an image such that the 3rd channel can be used as a depth map. Is a 1 in the depth map equivalent to 1 meter in the simulation (or some other unit)?

fxia22, 2022-04-18:
Yes, If you use the renderer, you should be able to use them directly as real-world units in meters.

Note if you use the depth sensors, it will be clipped and scaled to [0,1] by a upper bound and a lower bound defined in config files.

see:
https://github.com/StanfordVL/iGibson/blob/master/igibson/sensors/vision_sensor.py#L95-L108

CharlesAverill, 2022-04-21:
Got it, thank you @fxia22!

closed 2022-04-21 by CharlesAverill