StanfordVL/iGibson #198: Is VR supported on Linux systems?

JasonFengGit, 2022-04-21:
Hi there,
I am running igibson 2.1.0 (the lastest version) on a Ubuntu 20.04.1 machine. When trying to run `vr_simple_demo.py` under `igibson/examples/vr`. The program pops up this error, which I believe is because the Cmake doesn't compile the `VRRENDERCONTEXT`. 
[image]

Then I manully set USE_VR on in CMakeLists.txt to make CMake compile  `VRRENDERCONTEXT`, but `vr_simple_demo.py` still cannot run with another dependency error with CUDA.
[image]
[image]

Am I doing anything wrong?

By the way, we are trying to connect igibson with a `Oculus Quest 2` VR headset. It would be very helpful if you can give us some advice on how to do it.

Thank you so much!

mjlbach, 2022-04-21:
Yes, remove the build artifacts and install with `USE_VR=TRUE pip install -e .`, tested on ubuntu 20.04 with the latest nvidia driver (you need cuda on linux). I've tested with a rift 1, quest 2 should work with a link cable.

JasonFengGit, 2022-04-21:
Thank you so much for the quick reply! One thing i am a little confused with: for the `USE_VR=TRUE`, do you mean writing it in the CMakeLists.txt under igibson/render like I did, or doing it with something like `export USE_VR=TRUE`?

mjlbach, 2022-04-21:
Yep, cmake will transparently pick up the environmental variable. you can literally enter the command as I typed it. The cuda issue is likely either due to build artifacts or an old version of cuda on your system. I personally install cuda via cudatoolkit on the conda-forge conda channel and use 11.3

mjlbach, 2022-04-21:
Note the VR experience on linux is terrible on older nvidia drivers so make sure that is up to date.

JasonFengGit, 2022-04-21:
Thank you for the tips!

closed 2022-04-21 by JasonFengGit

HUSTColinLee, 2022-06-06:
> @mjlbach I encountered this when I install it with `USE_VR=TRUE pip install -e .`. Can you help with this issue? I am using CUDA11.4 and the latest driver. Many thanks! [image]

Solved by updating camke later than 3.11.0.