StanfordVL/iGibson #199: [DOESN'T EXIST] Examining Cubicasa5K scenes with igibson demo

ybkim95, 2022-04-21:
I followed the instruction from the following link to import Cubicasa5K dataset into igibson.
Link: https://github.com/StanfordVL/iGibson/tree/master/igibson/utils/data_utils/ext_scene

However, the code they refer doesn't exist.

To examine the scene, you can leverage the mouse_interaction.py in iGibson's example/demo folder [igibson/examples/demo/mouse_interaction.py](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/demo/mouse_interaction.py). 

Can someone help me with this?

roberto-martinmartin, 2022-04-21:
Hi @ybkim95 ,

sorry for the issue, the docs page is outdated, and the example they refer to doesn't exist anymore. One option is to go back into a version of iGibson (1.0) that still contains that demo and supports it.

Alternatively, you could you try modifying the example in `igibson/examples/scenes/ig_scene_selector` and manually set the path to the scene you want to load in the constructor of InteractiveIndoorScene. This would allow you to use the scenes with our latest iGibson version.
It has been a while since we tried those environments, we never tested them in iGibson 2.0, and there could be some issues, but they may work out-of-the-box.

Let me know how it goes.

--Roberto

ChengshuLi, 2022-06-10:
Feel free to re-open the issue if you have further questions.

closed 2022-06-10 by ChengshuLi