StanfordVL/iGibson #200: Intended operation of segmentation_to_rgb

CharlesAverill, 2022-05-03:
We're trying to convert segmentation masks into "did this object move?" masks. For example, if an object moved between frames, output an image that is black everywhere except for the new position of the moved object. This is what I've cooked up:

```python
def move_seg_processing(processed_ins_seg_frame, last_object_positions_orientations, new_object_positions_orientations, objects, colors_is):
    distance_threshold = 0.1
    out_img = np.empty(processed_ins_seg_frame.shape)
    
    for idx, obj in enumerate(objects):
        # If this object moved a big enough distance
        if distance(last_object_positions_orientations[idx][0], new_object_positions_orientations[idx][0]) > distance_threshold:
            # Go through each pixel in the instance segmentation mask
            for i in range(processed_ins_seg_frame.shape[0]):
                for j in range(processed_ins_seg_frame.shape[1]):
                    # If the color of this mask pixel matches the color at the same index of the object in question, make the output pixel white
                    if distance(processed_ins_seg_frame[i][j], colors_is[idx] * 255) < 0.1:
                        out_img[i][j] = 255

    return out_img
```

This does produce images where certain objects are highlighted, like so:
[image]
But it does not always highlight the intended object. Looking into the [segmentation_to_rgb](https://github.com/StanfordVL/iGibson/blob/216abff2e07a7d92f8f2d5e821ae9903ae54cf2b/igibson/utils/vision_utils.py#L74) function, I am not sure how the function is intended to operate.

Is there a way to map instance segmentation mask colors to object instances, to highlight them for our purposes?

cgokmen, 2022-05-03:
There is a way to map instance segmentation to objects. You can use this function on the renderer (accessible by doing `simulator.renderer`: https://github.com/StanfordVL/iGibson/blob/b1521122c1d39dbf392012b6fc0af99cbc62fcb7/igibson/render/mesh_renderer/mesh_renderer_cpu.py#L1929

It's pretty easy to use, see example use case:
https://github.com/StanfordVL/iGibson/blob/b1521122c1d39dbf392012b6fc0af99cbc62fcb7/igibson/object_states/robot_related_states.py#L134-L138

Since it is vectorized, if you provide it with the full segmentation image, it will produce a segmentation image where each pixel is colored by the corresponding body ID. You can then convert them to iGibson objects using `simulator.scene.objects_by_id[body_id]`

You shouldn't be getting the same body ID at the end of this for any two distinct objects. If that's still the case for the walls etc. please let us know.

Let us know if that helps.

CharlesAverill, 2022-05-03:
@cgokmen That was exactly what I needed, thank you so much!

closed 2022-05-03 by CharlesAverill