StanfordVL/iGibson #202: New ROS bridge compilation error

vbalaji21, 2022-05-12:
@roberto-martinmartin, You asked me to test the new ROS bridge and I did a fresh clone of the repo and tried to install it by compiling from the source method.
I was unable to install it and got error with pip install -e . command, I have attached the screenshot of the error log below.
[image]

[image]

roberto-martinmartin, 2022-05-12:
could it be that you didn't `git clone` recursively?

vbalaji21, 2022-05-19:
@roberto-martinmartin Thank you the problem was indeed I didn't do git clone recursively.

I have another doubt, I followed the instructions but I did not understand about the Python Path step as it has 4 paths and I don't know in which file I need to update them. Can you please let me know ?

I think I am facing this problem with the path (Python path).
I have attached the screenshot below.
[image]

vbalaji21, 2022-05-23:
@roberto-martinmartin can you please let me know