StanfordVL/iGibson #206: Probelms with getting vr_gaze_test.py to work

indoctrinatedrecluse, 2022-05-26:
Hello team!

I am facing issues getting `vr_gaze_test.py` to work. I am runnign iGibson on Ubuntu with an ALVR+Sidequest on an Oculus Quest 2 headset, and have had `vr_simple_demo.py` working so far. I have the `ig_dataset` added to an appropriate location which is updated in the global config yaml file. I have compiled from source using `USE_VR=TRUE pip install -e .`, and am trying to run `vr_gaze_test` using `python vr_gaze_test.py` while in the `igibson/examples/vr/` path.

I first got the following error:
[image]

Post this, I proceeded to change the parameter `renderer_params` on the following lines (68 and 92):
[image]
to `rendering_params` as expected by the `BaseObject` constructor in `object_base.py`. Re-running `python vr_gaze_test.py` gives me this error now:
[image]
which I believe refers to the following piece of code:
[image]
I cannot find an update() method as part of the `BehaviorRobot` class in `behavior_robot.py` and am unsure as to what needs to be called here. Therefore, I would like some help in getting this solved and `vr_gaze_test.py` working.

Thanks in advance for all the time, consideration and help!

cgokmen, 2022-05-26:
Hello,

We're sorry that this demo is causing issues! We will include the fix in our upcoming release. Meanwhile, you can fix this by applying the following changes:

1. replace `renderer_params` with `rendering_params`
2. replace `bvr_robot.update` with `bvr_robot.apply_action`
3. In `simulator_vr.py`, replace [this line](https://github.com/StanfordVL/iGibson/blob/58ac14cf62949008b6851a5a95602cd5084edffd/igibson/simulator_vr.py#L116) with `self.visual_object_cache = {}`

Let us know if that works!

Thanks,
Cem

cgokmen, 2022-05-26:
In fact, please simply clone the fix-gaze-demo branch of this repo - ignore the above fixes.

indoctrinatedrecluse, 2022-05-26:
Works now, thank you very much! PS: replacing the line in `simulator_vr.py` with `visual_object_cache = {}` was something I had already done to get `vr_simple_demo.py` working, good to know it was the right track! :D Thanks for all the help!

One further question:
When I run either `vr_simple_demo.py` or `vr_gaze_test.py`, the viewport remains still for the first run (on a newly built from source repo) and from the second run onwards, the viewport keeps advancing in a certain direction without any external input (from the controllers or headset). Any idea what could be causing this?

closed 2022-05-27 by indoctrinatedrecluse

cgokmen, 2022-05-27:
Sorry I just saw your response but I'm glad it's figured out!

HUSTColinLee, 2022-06-13:
@indoctrinatedrecluse Hi, I encountered the same issue. Could you please tell me how to fix it? I don't quite understand where to clone the fix-gaze-demo branch. Many thanks!

indoctrinatedrecluse, 2022-10-11:
Hey, with the updated changes, just a got switch fix-gaze-demo should work!