StanfordVL/iGibson #207: Run  examples on Mac (i9 install with pip)

PygmalionChen, 2022-06-03:
When I run the "python ./igibson/examples/robots/grasping_mode_example.py", the error occurs:
```
Traceback (most recent call last):
  File "/Users/cwz/opt/anaconda3/envs/ivln/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/cwz/opt/anaconda3/envs/ivln/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/examples/environments/env_int_example.py", line 8, in <module>
    from igibson.envs.igibson_env import iGibsonEnv
  File "/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/envs/igibson_env.py", line 13, in <module>
    from igibson.envs.env_base import BaseEnv
  File "/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/envs/env_base.py", line 8, in <module>
    from igibson.render.mesh_renderer.mesh_renderer_vr import VrSettings
  File "/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/render/mesh_renderer/mesh_renderer_vr.py", line 5, in <module>
    from igibson.render.mesh_renderer.mesh_renderer_cpu import MeshRenderer, MeshRendererSettings
  File "/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 13, in <module>
    from igibson.render.mesh_renderer import tinyobjloader
ImportError: cannot import name 'tinyobjloader' from 'igibson.render.mesh_renderer' (/Users/cwz/Home/PrjHome/PyPrj/PyTorch/IVLN-baselines/iGibson/igibson/render/mesh_renderer/__init__.py)
```

I wonder if the Mac version can't support the interactive render?

roberto-martinmartin, 2022-06-03:
the interactive renderer is supported on Mac. This looks more like a partial git clone. Did you follow the installation instructions and performed a git clone with `--recursive`? https://stanfordvl.github.io/iGibson/installation.html#compile-from-source

if not, do a post-clone recursive installation:
https://stackoverflow.com/questions/16773642/pull-git-submodules-after-cloning-project-from-github

Let us know if that solves your issues

PygmalionChen, 2022-06-04:
Thank you. Compile from source solves it!

roberto-martinmartin, 2022-06-05:
was your initial issue with a pip installation? if the dependencies are not correctly installed with pip, let us know, we will fix it.
thanks!

closed 2022-06-05 by roberto-martinmartin