StanfordVL/iGibson #211: Two video frames are blank when ubantu runs the example

THU-Bohr, 2022-06-07:
Hello developer, when I run the provided example, the robotview and viewer are blank, but the running data is normal and fps=30. What is the reason? Thank you。

fxia22, 2022-06-07:
can you tell us a bit more about your platform, linux / mac? nvidia driver version?

THU-Bohr, 2022-06-07:
ubantu 20.04 
RTX3060 nvidia driver version：30.0.14.7141
thank you

THU-Bohr, 2022-06-07:
ubantu 20.04
RTX3060 nvidia driver version：30.0.14.7141
thank you

从 Windows 版邮件发送

发件人: Fei Xia
发送时间: 2022年6月7日 9:48
收件人: StanfordVL/iGibson
抄送: THU-Bohr; Author
主题: Re: [StanfordVL/iGibson] Two video frames are blank when ubantu runsthe example (Issue #211)

can you tell us a bit more about your platform, linux / mac? nvidia driver version?