StanfordVL/iGibson #212: I have two  questions about the turtlebot view and pedestrain

Gengshucheng, 2022-06-08:
1、In the robot view, the above and below of the image are black, this may influence the learning,can I adjust it?
2、I want to import some Pedestrains in the scene, then how can I drive the Pedestrain with the social-forces model ORCA？
 Is there any tutorial for these two questions？
[image]
[image]

THU-Bohr, 2022-06-09:
Hello, which operating system did you use to install igibson? Which version? thank you

Gengshucheng, 2022-06-09:
> 
ubuntu20.04, igibson = 2.1.0

Gengshucheng, 2022-06-09:
> 

@THU-Bohr ubuntu20.04 and igibson=2.1.0

roberto-martinmartin, 2022-06-09:
@Gengshucheng , for the robot issue, try setting it to be hidden https://github.com/StanfordVL/iGibson/blob/master/igibson/render/mesh_renderer/mesh_renderer_settings.py#L27

The best way to set this parameter will depend on how you create your environment

For the use of simulated humans, I defer to some of my colleagues, I haven't use them much

Thanks!

roberto-martinmartin, 2022-06-09:
(in case it was not clear, the black stripes is the robot. these parts of the robot are visible in the images from the camera of the real robot because of its position, but if you are not planning to do sim2real, you may make the robot to be hidden for the camera with the rendering option I mentioned)

Gengshucheng, 2022-06-09:
@roberto-martinmartin  thanks，It's useful for me.

ChengshuLi, 2022-06-10:
@Gengshucheng For simulating pedestrians with ORCA, you can refer to this file:
https://github.com/StanfordVL/iGibson/blob/cvpr21_challenge/gibson2/tasks/social_nav_random_task.py

Make sure you install ORCA first: https://github.com/sybrenstuvel/Python-RVO2

closed 2022-06-10 by ChengshuLi