StanfordVL/iGibson #215: Impossible to replay BEHAVIOR VR demo

Roadsong, 2022-06-14:
There are basically two methods I tried to replay (as of June 14).

## Method 1
BEHAVIOR: master branch.
iGibson: master branch.
BDDL: master branch.
Dataset: version 2.0.6.

Command: `python -m igibson.examples.learning.demo_replaying_examples`
Results: I can see the demo, but the the motion of the robot is different from ground truth environment. I believe I faced a similar (or even worse) issue as https://github.com/StanfordVL/iGibson/issues/161#issuecomment-1022341060. I know it may because I am not running demos on a Windows machine, but it also shows the below messages and I don't know what's going on. I simply cannot find the `vr-demo-collection` branch at all. Also, the version 2.0.6 is suspicious, I don't know which version is correct for reproducing the demo.

```
********************************************************************************
WARNING:igibson.render.mesh_renderer.mesh_renderer_settings:WARN: Darwin does not support optimized renderer, automatically disabling
Warning, difference in git commits for repo: iGibson. This may impact deterministic replay
Logged git info:

{   'branch_name': 'vr-demo-collection',
    'code_diff': 'diff --git a/igibson/objects/multi_object_wrappers.py '
                 'b/igibson/objects/multi_object_wrappers.py\n'
                 'index 4827dad5..a3aa6610 100644\n'
                 '--- a/igibson/objects/multi_object_wrappers.py\n'
                 '+++ b/igibson/objects/multi_object_wrappers.py\n'
                 '@@ -129,7 +129,7 @@ class ObjectGrouper(BaseObject):\n'
                 ' \n'
                 '         # These attributes are used during object import '
                 'and should return\n'
                 '         # the concatenation results of all objects in '
                 'self.objects\n'
                 '-        if item in ["visual_mesh_to_material", '
                 '"link_name_to_vm", "body_ids", "is_fixed"]:\n'
                 '+        if item in ["visual_mesh_to_material", '
                 '"link_name_to_vm", "body_ids", "is_fixed", '
                 '"renderer_instances"]:\n'
                 '             return '
                 'list(itertools.chain.from_iterable(attrs))\n'
                 ' \n'
                 "         # Otherwise, check that it's the same for everyone "
                 'and then just return the value.\n'
                 '@@ -188,7 +188,6 @@ class ObjectGrouper(BaseObject):\n'
                 '             if issubclass(state_type, '
                 'AbsoluteObjectState):\n'
                 '                 '
                 'state_instance.load(dump[get_state_name(state_type)])\n'
                 ' \n'
                 '-\n'
                 ' class ObjectMultiplexer(BaseObject):\n'
                 '     """A multi-object wrapper that acts as a proxy for the '
                 'selected one between the set of objects it contains."""\n'
                 ' \n'
                 'diff --git '
                 'a/igibson/render/mesh_renderer/mesh_renderer_vr.py '
                 'b/igibson/render/mesh_renderer/mesh_renderer_vr.py\n'
                 'index a768ad31..f766f0ac 100644\n'
                 '--- a/igibson/render/mesh_renderer/mesh_renderer_vr.py\n'
                 '+++ b/igibson/render/mesh_renderer/mesh_renderer_vr.py\n'
                 '@@ -138,7 +138,7 @@ class VrSettings(object):\n'
                 '         self.use_tracked_body = '
                 'shared_settings["use_tracked_body"]\n'
                 '         self.torso_tracker_serial = '
                 'shared_settings["torso_tracker_serial"]\n'
                 '         # Both body-related values need to be set in order '
                 'to use the torso-tracked body\n'
                 '-        self.using_tracked_body = self.use_tracked_body and '
                 'self.torso_tracker_serial\n'
                 '+        self.using_tracked_body = self.use_tracked_body and '
                 'bool(self.torso_tracker_serial)\n'
                 '         if self.torso_tracker_serial == "":\n'
                 '             self.torso_tracker_serial = None\n'
                 ' \n'
                 'diff --git a/igibson/robots/behavior_robot.py '
                 'b/igibson/robots/behavior_robot.py\n'
                 'index bc984df9..be99fa9f 100644\n'
                 '--- a/igibson/robots/behavior_robot.py\n'
                 '+++ b/igibson/robots/behavior_robot.py\n'
                 '@@ -195,6 +195,8 @@ class BehaviorRobot(ManipulationRobot, '
                 'LocomotionRobot, ActiveCameraRobot):\n'
                 ' \n'
                 '         # TODO: Remove hacky fix - constructor/config '
                 'should contain this data.\n'
                 '         if self.simulator.mode == SimulatorMode.VR:\n'
                 '+            print("robot:", self.use_tracked_body)\n'
                 '+            print("sim:", '
                 'self.simulator.vr_settings.using_tracked_body)\n'
                 '             assert (\n'
                 '                 self.use_tracked_body == '
                 'self.simulator.vr_settings.using_tracked_body\n'
                 '             ), "Robot and VR config do not match in terms '
                 'of whether to use tracked body. Please update either '
                 'config."\n'
                 'diff --git a/igibson/vr_config.yaml '
                 'b/igibson/vr_config.yaml\n'
                 'index b6051117..13b7007b 100644\n'
                 '--- a/igibson/vr_config.yaml\n'
                 '+++ b/igibson/vr_config.yaml\n'
                 '@@ -36,7 +36,7 @@ shared_settings:\n'
                 '   # Serial number of VR torso tracker - this can be found '
                 'by connecting/pairing the tracker,\n'
                 '   # then going into Steam VR settings -> controllers -> '
                 'manage vive trackers\n'
                 '   # Note: replace this with your own tracker serial number '
                 'or leave blank to not use one\n'
                 '-  torso_tracker_serial: "LHR-DF82C682"\n'
                 '+  torso_tracker_serial: "LHR-BDE12AB6"\n'
                 ' # Settings that are specific to different VR devices (eg. '
                 'eye tracking, button mapping)\n'
                 ' device_settings:\n'
                 '   HTC_VIVE_PRO_EYE:',
    'code_diff_staged': '',
    'commit_hash': 'bc2520de66025c486cff11e30d881b0f29cd1384'}
Current git info:

{   'branch_name': 'master',
    'code_diff': 'diff --git a/igibson/tasks/behavior_task.py '
                 'b/igibson/tasks/behavior_task.py\n'
                 'index 3c1b9868..d38b763f 100644\n'
                 '--- a/igibson/tasks/behavior_task.py\n'
                 '+++ b/igibson/tasks/behavior_task.py\n'
                 '@@ -127,6 +127,12 @@ class BehaviorTask(BaseTask):\n'
                 '             self.conds, self.backend, self.object_scope, '
                 'self.goal_conditions\n'
                 '         )\n'
                 ' \n'
                 "+        # print('[DEBUG]][self.obj_scope]', "
                 'self.object_scope)\n'
                 '+        # print(self.initial_conditions)\n'
                 '+        # print(self.goal_conditions)\n'
                 "+        # print('[DEBUG][self.ground_goal_state_options]', "
                 'self.ground_goal_state_options)\n'
                 '+        # exit()\n'
                 '+\n'
                 '         # Demo attributes\n'
                 '         self.instruction_order = '
                 'np.arange(len(self.conds.parsed_goal_conditions))\n'
                 '         np.random.shuffle(self.instruction_order)',
    'code_diff_staged': '',
    'commit_hash': '58ac14cf62949008b6851a5a95602cd5084edffd'}
Creating environment and resetting it
```

## Method 2
BEHAVIOR: master branch.
iGibson: behavior-replay branch.
BDDL: master branch / behavior-refactored-verified-problems.
Dataset: version 2.0.6.

Command: `python -m igibson.examples.behavior.behavior_demo_replay`
Results: Error, no module named 'bddl.activity_base'. / Error, no module named 'igibson.task'

## Summary
In short, impossible to replay at all. Can anyone write a detailed instructions on that? If you want the benchmark get attention, at least people can reproduce the basic things easily, right? Many instructions are not up-to-date and show many inconsistencies. Can anyone tells me which one is the correct one to reproduce the vr demo? This https://github.com/StanfordVL/behavior/blob/main/docs/vr_demos.md does not work.

cgokmen, 2022-06-14:
Can you try switching all three repos (BEHAVIOR, iGibson, BDDL) to the behavior_replay branch and try again?

cgokmen, 2022-06-14:
My bad - you don't need toe BEHAVIOR repo for this. Just the iGibson repo (behavior-replay branch) and the BDDL repo (also behavior-replay). The replay script will be in examples/behavior/behavior_demo_replay I think.

cgokmen, 2022-06-14:
You also need the behavior-replay copy of ig_dataset (https://storage.googleapis.com/gibson_scenes/ig_dataset_replay.zip) and of the robot assets (https://github.com/StanfordVL/ig_assets/tree/behavior-replay).

So essentially you need all four components to be behavior_replay copies:

* iGibson: get from github
* BDDL: get from github
* ig_dataset: get from above URL
* assets: get from github

Roadsong, 2022-06-15:
Hi @cgokmen, thanks for you quick reply. I basically followed your advice and installed suggested version/branches, etc.

## Exp 1
BEHAVIOR: null
iGibson: behavior-replay branch.
BDDL: behavior_refactor_verified_problems branch.
Dataset version: v2.0.1, downloaded.
Assets: downloaded.

Installed in a conda environment, created by `conda create -n replay python=3.7`
Install iGibson by `pip install -e ./iGibson`
Install BDDL by `python setup.py install` (behavior_refactor_verified_problems branch)

Command: `python -m igibson.examples.behavior.behavior_demo_replay`
Results: 
```
Traceback (most recent call last):
  File "/Users/xxx/opt/anaconda3/envs/replay/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/xxx/opt/anaconda3/envs/replay/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/xxx/Desktop/research/replay/iGibson/igibson/examples/behavior/behavior_demo_replay.py", line 280, in <module>
    main()
  File "/Users/xxx/Desktop/research/replay/iGibson/igibson/examples/behavior/behavior_demo_replay.py", line 268, in main
    bddl.set_backend("iGibson")
  File "/Users/xxx/Desktop/research/replay/bddl/bddl/__init__.py", line 10, in set_backend
    from igibson.task.bddl_backend import IGibsonBDDLBackend
ModuleNotFoundError: No module named 'igibson.task'
```

It seems that the iGibson's `behavior-replay` branch is not compatible with something else. I do find `igibson.task` code in the main branch and I am not sure if the problem can be solve if I manually include those code into the current `behavior-replay` branch. 

Something else: Also, I simply find that in BDDL repo `behavior_refactor_verified_problems` branch, the `pytest` **cannot pass**. As of https://github.com/StanfordVL/bddl/issues/10#issuecomment-1022554441 mentioned, a correct branch/commit of BDDL is really hard to find. 

## Exp2
If I somehow changed the BDDL branch to master branch, the `behavior-replay` branch of iGibson code does not work because 
```
Traceback (most recent call last):
  File "/Users/xxx/opt/anaconda3/envs/replay/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/xxx/opt/anaconda3/envs/replay/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/xxx/Desktop/research/replay/iGibson/igibson/examples/behavior/behavior_demo_replay.py", line 15, in <module>
    from igibson.activity.activity_base import iGBEHAVIORActivityInstance
  File "/Users/xxx/Desktop/research/replay/iGibson/igibson/activity/activity_base.py", line 6, in <module>
    from bddl.activity_base import BEHAVIORActivityInstance
ModuleNotFoundError: No module named 'bddl.activity_base'
```

It seems that the `master` branch code of `iGibson` and `BDDL` works better (at least I can see the windows).

cgokmen, 2022-06-18:
I have pushed a behavior-replay branch onto BDDL, can you try using that? Thanks.

Roadsong, 2022-06-21:
Hi @cgokmen, thanks for your time and effort.

I basically somehow managed to reproduce the demo on a mac machine using the `behavior-replay` branch of iGibson and `behavior_verfied_problems` branch of BDDL. Surely I include the missing code and files, and I also did some necessary modifications to make it work. I am glad to see a demo running on a window, e.g., cleaning_windows, even the results are not successful. However, it seems that for deterministic replay (exact match of every single steps), I must get everything work on a windows machine, since the `pybullet-svl` library may yield different results across platforms (as mentioned in https://github.com/StanfordVL/bddl/issues/10#issuecomment-1022562103)

So, I tried to reproduce this in a **virtual** win 10 machine to see if the full deterministic replay is possible. That was not easy because I was using a **virtual** win 10 machine (to be specific, the host os is MacOS, and the guest OS is Win10 running on Parallel Desktop, the graphic card is AMD series). After fixing some issues and installing a good vs compiler, I could successfully build and install the iGibson, and BDDL seemed to work (I tried `behavior-replay` branch this time). 

However, I got the following problem this time:

```
ERROR: Failed to create GLFW window.
```

Well, I found this error was trigger by https://github.com/StanfordVL/iGibson/blob/58ac14cf62949008b6851a5a95602cd5084edffd/igibson/render/cpp/glfw_mesh_renderer.cpp#L91
because `glfwCreateWindow` was not successful.

I googled a lot about this problem, especially paid attention to some platform-specific questions and answers. The results were fixed and I suspect that the `glfw` cannot (easily) work on a win 10 guest machine running on parallel desktop. It looks like the OpenGL support from parallel desktop is not good enough, yet I was not sure about this.

My question is, can I at least do some test on a virtual windows machine without Nvidia graphic cards, CUDA, etc? It seems that the MacOS without Nvidia and CUDA seems fine except the determinism problem. Or, I must have a physical win 10 machine and satisfy the requirements https://stanfordvl.github.io/iGibson/installation.html#system-requirements?

cgokmen, 2022-06-25:
As far as I know you do need a Windows machine with a GPU, but let me defer to @fxia22 who will know these requirements better. Perhaps you could get one on Azure or something? Even a somewhat older GPU like a 1080 will work fine for this purpose. Sorry I can't be more helpful.

Roadsong, 2022-06-29:
Hi @cgokmen, thanks for your time and effort again, you really helped a lot! You don't need to answer the following questions, I just want to update the issue in case anyone else has similar problems.

Basically I used a physical windows machine (without GPU, but I don't think GPU is the cause of following problems). I debug and read the source code carefully and found that I still couldn't reproduce an activity deterministically.

## Experiment
- iGibson: `behavior-replay` branch
- BDDL: 'behavior-repaly` branch
- ig_dataset: version `v2.0.1`
- asset: version `v2.0.6`
- HDF5 file: `bottling_fruit_0_Wainscott_0_int_0_2021-05-24_19-46-46.hdf5` can be downloaded at this [link](https://download.cs.stanford.edu/downloads/behavior/bottling_fruit_0_Wainscott_0_int_0_2021-05-24_19-46-46.hdf5)
- Script: ```python -m igibson.examples.behavior.behavior_demo_replay --mode="headless" --vr_log_path="xxx.hdf5"```

## Logs, cannot replay deterministically
```
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 200 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 400 -----
----- Writing log data to hd5 on frame: 600 -----
----- Writing log data to hd5 on frame: 800 -----
----- Writing log data to hd5 on frame: 1000 -----
----- Writing log data to hd5 on frame: 1200 -----
----- Writing log data to hd5 on frame: 1400 -----
----- Writing log data to hd5 on frame: 1600 -----
----- Writing log data to hd5 on frame: 1800 -----
----- Writing log data to hd5 on frame: 2000 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 2200 -----
----- Writing log data to hd5 on frame: 2400 -----
----- Writing log data to hd5 on frame: 2600 -----
----- Writing log data to hd5 on frame: 2800 -----
----- Writing log data to hd5 on frame: 3000 -----
Demo was succesfully completed:  False
IG LOGGER INFO: Ending log writing session after 3000 frames
Mismatch for obj 120 with mismatched attribute joint_state starting at timestep 443
Mismatch for obj 163 with mismatched attribute joint_state starting at timestep 1
Mismatch for obj 163 with mismatched attribute orientation starting at timestep 1
Mismatch for obj 163 with mismatched attribute position starting at timestep 1538
Mismatch for obj 164 with mismatched attribute joint_state starting at timestep 2
Mismatch for obj 164 with mismatched attribute orientation starting at timestep 2
Mismatch for obj 164 with mismatched attribute position starting at timestep 1654
Mismatch for obj 165 with mismatched attribute orientation starting at timestep 2
Mismatch for obj 165 with mismatched attribute position starting at timestep 1515
Mismatch for obj 199 with mismatched attribute orientation starting at timestep 577
Mismatch for obj 199 with mismatched attribute position starting at timestep 606
Mismatch for obj 202 with mismatched attribute orientation starting at timestep 919
Mismatch for obj 202 with mismatched attribute position starting at timestep 919
Mismatch for obj 270 with mismatched attribute orientation starting at timestep 0
Mismatch for obj 270 with mismatched attribute position starting at timestep 0
Demo was deterministic:  False
```
I was really concerned about the obj 270, since mismatch started at timestep 0,  and it proved to be the Behavior Robot itself.

## Information recorded in the HDF5 file
- From my understanding, the HDF5 file (~8MB) includes the robot's actions, and other information of the relevant object states, poses, etc. I checked out the structure of the HDF5 file and I could see following data groups.
 ``` 
'/', '/agent_actions', '/agent_actions/vr_robot', 
'/frame_data', 
'/goal_status', '/goal_status/satisfied', '/goal_status/unsatisfied', 
'/physics_data',
'/physics_data/1', '/physics_data/1/joint_state', '/physics_data/1/orientation', '/physics_data/1/position', 
'/physics_data/103', '/physics_data/103/joint_state', '/physics_data/103/orientation', '/physics_data/103/position', 
'/physics_data/107', '/physics_data/107/joint_state', '/physics_data/107/orientation', '/physics_data/107/position', 
'/physics_data/120', '/physics_data/120/joint_state', '/physics_data/120/orientation', '/physics_data/120/position', 
'/physics_data/163', '/physics_data/163/joint_state', '/physics_data/163/orientation', '/physics_data/163/position', 
'/physics_data/164', '/physics_data/164/joint_state', '/physics_data/164/orientation', '/physics_data/164/position', 
'/physics_data/165', '/physics_data/165/joint_state', '/physics_data/165/orientation', '/physics_data/165/position', 
'/physics_data/199', '/physics_data/199/joint_state', '/physics_data/199/orientation', '/physics_data/199/position', 
'/physics_data/202', '/physics_data/202/joint_state', '/physics_data/202/orientation', '/physics_data/202/position', 
'/physics_data/270', '/physics_data/270/joint_state', '/physics_data/270/orientation', '/physics_data/270/position', 
[... cut ...]
- This time, the result is `'BRBody_1': {'xyz': array([3.20019555, 8.90034866, 0.49919999]), 'rpy': array([-7.12756938e-04,  3.87323983e-04, -8.18337575e-05])}` and the position (xyz) is slightly different from what we saw in the HDF5 file.

## Solution
- I was not sure if the version of the assets may affect this. Since `v2.0.6` and `v2.0.1` are different. Do we have `v2.0.1` of assets that may have slight differences?
- The link to download assets: [link](https://storage.googleapis.com/gibson_scenes/assets_igibson.tar.gz), I can confirm it is `v2.0.6`, maybe the older version has been overwritten by the new one. But I am not so sure. @fxia22 do you have any opinions?

loubron23, 2024-09-27:
> Hi @cgokmen, thanks for your time and effort again, you really helped a lot! You don't need to answer the following questions, I just want to update the issue in case anyone else has similar [...]

hello
after 2 years, I‘m no facing the same problem:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
eyeINFO:root:Loading /home/loubron/disk/BEHAVIOR/iGibson/igibson/data/assets/models/vr_agent/vr_eye/normal_color/vr_head.obj
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 200 -----
----- Writing log data to hd5 on frame: 400 -----
----- Writing log data to hd5 on frame: 600 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 800 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 1000 -----
----- Writing log data to hd5 on frame: 1200 -----
----- Writing log data to hd5 on frame: 1400 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 1600 -----
----- Writing log data to hd5 on frame: 1800 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 2000 -----
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
WARNING:root:Trying to set hidden state before vertices are merged, converted to no-op
----- Writing log data to hd5 on frame: 2200 -----
----- Writing log data to hd5 on frame: 2400 -----
Demo was succesfully completed:  False
IG LOGGER INFO: Ending log writing session after 2400 frames
Mismatch for obj 169 with mismatched attribute orientation starting at timestep 1305
Mismatch for obj 169 with mismatched attribute position starting at timestep 1412
Mismatch for obj 45 with mismatched attribute joint_state starting at timestep 1178
Mismatch for obj 62 with mismatched attribute orientation starting at timestep 1581
Mismatch for obj 62 with mismatched attribute position starting at timestep 1581
Mismatch for obj 63 with mismatched attribute orientation starting at timestep 1279
Mismatch for obj 63 with mismatched attribute position starting at timestep 1279
Mismatch for obj 64 with mismatched attribute orientation starting at timestep 9
Mismatch for obj 64 with mismatched attribute position starting at timestep 1788
Mismatch for obj 65 with mismatched attribute orientation starting at timestep 16
Mismatch for obj 65 with mismatched attribute position starting at timestep 1765
Mismatch for obj 66 with mismatched attribute orientation starting at timestep 27
Mismatch for obj 66 with mismatched attribute position starting at timestep 1333
Mismatch for obj 67 with mismatched attribute orientation starting at timestep 27
Mismatch for obj 67 with mismatched attribute position starting at timestep 1545
Mismatch for obj 68 with mismatched attribute orientation starting at timestep 2115
Mismatch for obj 68 with mismatched attribute position starting at timestep 2115
Mismatch for obj 69 with mismatched attribute orientation starting at timestep 27
Mismatch for obj 69 with mismatched attribute position starting at timestep 1177
Demo was deterministic:  False

Could you please help me? Really appreciate！