StanfordVL/iGibson #219: [MoMart] task definitions / environments

dHonerkamp, 2022-06-21:
Hi, I am interested in the tasks introduced in the MoMart paper (https://sites.google.com/view/il-for-mm/home?authuser=0)

While the website decribes how to download the dataset, it does not mention anything on how to test on the 5 tasks that they introduce for their benchmark. I've been looking at the momart branch as detailed on the website, but it is very hard to figure out which aspects belong to MoMart.

Would it be possible to provide the commands that I need to use to evaluate these five tasks (Table Cleanup to Dishwasher, Table Cleanup to Sink, Table Setup from Dresser, Table Setup from Dishwasher, Unload Dishwasher to Dresser), ideally both in the standard version and in the randomized version used for the "Few-Shot Generalization" section of the paper?

Your help is much appreciated.

dHonerkamp, 2022-06-24:
@cremebrule @andreykurenkov do you maybe have a pointer on this?

jehanyang, 2023-11-29:
Would appreciate a follow-up on this. I'm interested in evaluating these tasks as well with the MoMart dataset.