StanfordVL/iGibson #222: If I am allowed to publish a dataset derived from the old Gibson dataset?

chengzhag, 2022-07-04:
Dear authors @b0ku1, @fxia22, @ChengshuLi, @roberto-martinmartin  of iGibson 1.0:

I am the first author of the ICCV 2021 paper _DeepPanoContext: Panoramic 3D Scene Understanding with Holistic Scene Context Graph and Relation-based Optimization_ ([home page](https://chengzhag.github.io/publication/dpc/)).

I used your iGibson 1.0 package and the dateset for my paper. At the time, the dataset is unencrypted, so I generated a new dataset with rendered images, formatted layout and object information, and watertight object models for my project. After the paper was published, the code for dataset generation is released, but the generated data is not.

However, after iGibson 2.0 comes out, only encrypted data is provided, and the updated license prohibits extracting any Data from iGibson:
[image]
This makes my data generation code unable to be reproduced for new users (https://github.com/chengzhag/DeepPanoContext/issues/5).

Although I am not allowed to distribute the unencrypted data as the license said. But I would like to ask if I am allowed to publish my generated dataset derived from the old iGibson data, since it looks like to be allowed in the old license:
[image]

I will greatly appreciate your response.

Thank you,
Cheng Zhang

ChengshuLi, 2022-08-16:
@chengzhag 

Apologies for my belated reply.

If your dataset is derived from the iGibson 1.0 dataset only, feel free to release it. The reason we need to encrypt the current dataset is that many objects from iGibson 2.0 are not allowed to be used outside of iGibson due to licensing restrictions.

chengzhag, 2022-08-17:
Thank you @ChengshuLi for your response, really appreciate it.

closed 2022-08-17 by chengzhag

fxia22, 2022-08-17:
Cool paper by the way, congrats!