StanfordVL/iGibson #225: camera set problem

Zhaocka, 2022-07-08:
Hello, 
I want to set different camera heights to get different observations, but I haven't found the relevant interface?

Cyclohexene, 2022-09-04:
Having the same issue here, is there any way to set camera poses?

stalhabukhari, 2022-10-12:
@Zhaocka @Cyclohexene interested in knowing if there has been any update to this.

roberto-martinmartin, 2022-10-12:
Hi,
are you talking about the camera on the agent or the external viewer? 
The external viewer can be changed with mouse interactions, or programmatically.
To change the camera on the agent, you need to change the location of the frame "eyes" (I believe) in the URDF file of the robot you are using. 
@Cyclohexene do not worry about the warning. That is when used within ROS: the file will be automatically generated. But within iGibson, the file won't change externally, your changes will be permanent.

If someone implements the functionality to define an additional offset in the config file to be added to the camera with respect to the "eyes" frame in the URDF, that would be fantastic! Please, create a PR for it.

ConnerQiu, 2023-04-06:
@Cyclohexene great that you have solve this problem! Would you like to share your modification to me? since I have the same need to add different cameras on the robot. You sharing would be great help for me. Thanks!!