StanfordVL/iGibson #229: Large 3D reconstructed world import

JasonLiu324, 2022-08-09:
Hi guys, our team attends to import large 3D reconstructed world (larger than 1 G) and the texture file is not easy to be combined into one texture file. So I wonder that have you considered or tried to import large 3D world into iGibson? Thank you！

JasonLiu324, 2022-08-12:
OK, I have imported a large obj file (more than 800M) successfully in static scene, but the FPS is relatively slow (3~4). And when I failed to add robot in it or integrate it with ROS with exception of segment fault (core dumped). So I suppose that iGibson cannot handle large 3D constructed world.

ChengshuLi, 2022-08-16:
@fxia22 Any ideas?
I would suggest maybe modifying the `texture_scale` to `0.5` and see if there is any speedup?

JasonLiu324, 2022-08-18:
I have tired reducing the texture_scale to 0.5, and the FPS rises from 3-4  to 4-5.

fxia22, 2022-08-23:
How large is the space itself? We have reconstructed large spaces and manage to represent it efficiently with a small 3d file. You can consider doing some mesh decimation to make it smaller, blender would be an excellent tool for that. [tutorial](https://www.youtube.com/watch?v=Yx9TvvnxCAM)

fxia22, 2022-08-23:
I suspect the problem is not texture size, but it is caused by obj file with too many polygons.

JasonLiu324, 2022-08-26:
@fxia22 Thanks for your reply! The space itself is about 500-600 square meters. We have tried downsampling using Blender to reduce the .obj file size, but it will cause some empty space which may harm robot movement. The vedio you provide is really helpful and I assume making such adjustment is very time-consuming lol.

closed 2022-09-15 by JasonLiu324