StanfordVL/iGibson #230: Broken link to iGibson Stanford 2D-3D-S dataset

joelloo, 2022-08-14:
Hello,

I would like to use the Stanford 2D-3D-S dataset in iGibson. The link for the iGibson 2D-3D-S dataset that I received after submitting the end-user agreement was invalid. The link specified in the email I received is at the directory gibsonchallenge/2d3ds_for_igibson.zip, but this yielded a NoSuchKey error. The link to the older _Gibson_ 2D-3D-S dataset remains valid however. Can I check if the iGibson 2D-3D-S dataset has been removed for some reason, and if it can be made available again?

Thanks!

fxia22, 2022-08-16:
We are working on restoring the access

ChengshuLi, 2022-08-16:
@joelloo Thanks for letting us know. 

We have restored the dataset at this URL location: https://storage.googleapis.com/gibsonchallenge/2d3ds_for_igibson.zip
Once you unzip it, you will see a folder called `2d3ds_v3` that contains the data.

Feel free to re-open the issue if this doesn't work. Thanks!

closed 2022-08-16 by ChengshuLi