StanfordVL/iGibson #232: Missing Demo File for iGibson ROS rs.yaml

GaganBhat, 2022-08-23:
Looks like the map folder from igibson-ros is missing inside the repository. 

This leads to a demo failing from the documentation. 

Steps to reproduce ([instructions from here](https://stanfordvl.github.io/iGibson/ros_integration.html#running)): 

1. Setup igibson-ros, download assets and demo dataset as needed.
2. Run `roslaunch igibson-ros turtlebot_navigation.launch`. 

Error when run: 

`[ERROR] [1661213552.429941046]: Map_server could not open /home/gaganb/catkin_ws/src/igibson-ros/map/rs.yaml.`

Leads to a broken navigation stack demo for turtlebot with igibson-ros.

JasonLiu324, 2022-08-23:
@GaganBhat I have happened to solve this question. You should use the gmapping launch file and teleoperate the turtlebot to get a map first (rosrun map_server map_saver) and then copy the map files into /igibson-ros/map/. Then it will be solved.

fxia22, 2022-08-23:
@JasonLiu324 thanks, that's right. you need to run mapping and `map_saver` first

fxia22, 2022-08-23:
If you guys generated one, feel free to file a pull request (with both the map image file and yaml file). Thanks.

closed 2022-09-08 by GaganBhat