StanfordVL/iGibson #233: ROS Odometry Calculation -- Magic Number Clarification

GaganBhat, 2022-08-23:
I was looking to get more insight into how this magic number inside of `turtlebot_rgbd.py` was calculated. 

Specifically, [this line](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/ros/igibson-ros/turtlebot_rgbd.py#L145):

`odom_msg.twist.twist.angular.z = (self.cmdy - self.cmdx) * 5 * **8.695652173913043**`

What does **8.695652173913043** represent? 

I'm assuming it is something to do with wheel base of the robot, but unsure.

fxia22, 2022-08-23:
Sorry for the confusion, cmdx and cmdy are the angular velocity for the two wheels, here we want to calculate the angular velocity of the vehicle. 

It should be (cmdy - cmdx) * wheel radius / wheel base.

GaganBhat, 2022-08-23:
If it is wheel radius **/** wheel base, why is 5 (the wheel radius I am assuming) being _multiplied_ with 8.69 (the wheel base)? @fxia22

kamalesh0406, 2022-08-23:
Since we are setting the linear and angular velocity, shouldn't the code be 
```
odom_msg.twist.twist.linear.x = self.cmdx
odom_msg.twist.twist.angular.z = -self.cmdy
```
since `self.cmdx` is the linear velocity and `self.cmdy` is the angular velocity according to the rest of the code.