StanfordVL/iGibson #234: Distance from an object to nearest wall

Zesheng-Xu, 2022-08-23:
Hi, I am working on setting up scene objects for my project. I am running into a problem where often an object is placed too close to a wall(Not in a wall), thus I am wondering if there is a way to calculate the distance from the object to the wall, thank you!

sycz00, 2022-08-24:
I think there does not exist a method for the distance but you could take a look into the function for sampling spawn positions. As far as I remember, it is based on a static traverseability map (either fully packed with all objects or without any object). So for example, by inflating the map more and more, you sample less points close to walls. Another (dirty) way is to sample points around your spawn position and observe any collisions with the wall (reject spawns based on that).

Zesheng-Xu, 2022-08-25:
Thank you for your response, we are already using the sample spawn and collision points checking method, at this point, I think inflating the map will be the only choice left for us.

closed 2022-08-25 by Zesheng-Xu