StanfordVL/iGibson #237: Robot / Map Pose Glitches and Shifts during Rotation in ROS

GaganBhat, 2022-09-08:
I noticed an issue when using igibson-ros `turtlebot_gmapping` example. 

When the robot rotates in place, the pose of the robot + the map around it teleports back and forth substantially. 

https://user-images.githubusercontent.com/25236969/189164726-6c5be578-cd9d-47e1-9472-83f59795817b.mp4

In the simulator it is spinning in place, but the odometry that is calculated by the igibson script rapidly jumps around.

Any ideas what the issue could be?

Here is a video showing this issue, notice how the robot / costmap jumps around in space.

JasonLiu324, 2022-09-15:
Hi, this is because the speed of turtlebot is too fast. Maybe you can try to slow down the speed of turtlebot.

GaganBhat, 2022-09-15:
This seems to happen regardless of the speed of the turtlebot. 

In a simulation environment, since we know the ground truth, how could these types of localization issues occur?

[image]