StanfordVL/iGibson #238: Is the 2d3ds_for_igibson dataset broken?

SophieLeichtle, 2022-09-15:
I've been trying to look at a variety of the available datasets as i need a framework for training a robot from visuals, and while a lot of scenes have a lot of artifacting the 2d3ds datasets seem outright broken.
[image]
[image]
[image]

These screenshots are just from area1, but the other areas have similar issues. Is this a problem with the dataset or could there be something wrong with my installation?

ChengshuLi, 2022-09-15:
@fxia22 Correct me if I am wrong, but I think this is expected behavior? 3d reconstruction is not perfect and 2d3ds dataset is from a few years ago: the local artifacts seem to be expected.

SophieLeichtle, 2022-09-15:
makes sense, the artifacts just seem so strong that it doesnt really look like its very usable. just want to make sure its not something wrong with my installation to be safe

closed 2022-09-29 by SophieLeichtle