StanfordVL/iGibson #243: feature or bug: IndoorScene.map_to_world and .world_to_map NOT inverse to each other

yihe-61wu, 2022-10-20:
I've become confused when starting to visualise the traversability map. I believe (might be wrong, though) the problem comes from the inconsistency in these two methods of `IndoorScene`:
```
    def map_to_world(self, xy):
        """
        Transforms a 2D point in map reference frame into world (simulator) reference frame

        :param xy: 2D location in map reference frame (image)
        :return: 2D location in world reference frame (metric)
        """
        axis = 0 if len(xy.shape) == 1 else 1
        return np.flip((xy - self.trav_map_size / 2.0) * self.trav_map_resolution, axis=axis)

    def world_to_map(self, xy):
        """
        Transforms a 2D point in world (simulator) reference frame into map reference frame

        :param xy: 2D location in world reference frame (metric)
        :return: 2D location in map reference frame (image)
        """
        return np.flip((np.array(xy) / self.trav_map_resolution + self.trav_map_size / 2.0)).astype(np.int)
```
Firstly, I don't understand why `np.flip` is needed. My guess is that perhaps numpy arrays indexing elements differently from a .png file (provided for ig_scenes) indexing pixels. Please enlighten me!

Secondly, why one method flip along a specific axis, yet the other flip all axes?

Thank you!

ChengshuLi, 2022-11-15:
You are right that the two functions are not exactly the inverse of each other. But the logic is the same. It's just that in `map_to_world`, we allow additional flexibility of xy having the shape of `[N, 2]` or `[2]`. 

`np.flip` is needed because the world and the map have their x-y axis flipped. 

Feel free to re-open the issue if you have any additional questions.

closed 2022-11-15 by ChengshuLi

yihe-61wu, 2022-11-15:
Thank you for the clear explanation.

I have one follow-up question: Naively, I thought after transforming the coordinates, I could map them directly onto the traversability map in the .png file (provided for ig_scenes). However, I found there were some offsets in the coordinates. 

To rephrase my question: My goal is to draw the trajectory of a robot on the traversability map. After obtaining (and transforming) the coordinates, how can I draw it on the .png file? Are there some other transformations required? Or perhaps I shouldn't use `IndoorScene.map_to_world` and `.world_to_map` at all?

I believe I must have mistaken something, but I don't know what. Thank you in advance!

ChengshuLi, 2022-11-30:
Yes, you should be able to map them directly onto the traversability map using `.world_to_map`. The map coordinates correspond to the [row, col] coordinate of the numpy array of the .png file.