StanfordVL/iGibson #245: Building wheel for igibson (pyproject.toml) ... error

Vaibhavmulti, 2022-10-30:
I am getting these erros while installing igibson. I have also attached the text file for entire log.
Please let me know how to proceed. 

Building wheel for igibson (pyproject.toml) did not run successfully.
  │ exit code: 1

error: could not create 'build\lib.win-amd64-cpython-310\igibson\examples\ros\igibson-ros\turtlebot\turtlebot_description\meshes\stacks\circles\68-02421-8000-RA_Turtlebot_F-F_Standoff_color.png': No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for igibson

ERROR: Could not build wheels for igibson, which is required to install pyproject.toml-based projects

[log.txt](https://github.com/StanfordVL/iGibson/files/9896120/log.txt)

cgokmen, 2022-10-30:
Can you try cloning the git repository, going into its directory, and running `pip install -e .` and report the result of that?

Vaibhavmulti, 2022-10-31:
[log.txt](https://github.com/StanfordVL/iGibson/files/9897452/log.txt)

Xqle, 2022-10-31:
Try adding `SETUPTOOLS_ENABLE_FEATURES="legacy-editable"` to your system environment variables. It works when I ran into the same issue as yours.

Vaibhavmulti, 2022-10-31:
[image]
I tried setting the system environment variable still not working.

[logpipinstalligibson.txt](https://github.com/StanfordVL/iGibson/files/9899128/logpipinstalligibson.txt)
[logpipinstall-e.txt](https://github.com/StanfordVL/iGibson/files/9899157/logpipinstall-e.txt)

fxia22, 2022-10-31:
CMake Error at CMakeLists.txt:2 (project):
      No CMAKE_C_COMPILER could be found.

    CMake Error at CMakeLists.txt:2 (project):
      No CMAKE_CXX_COMPILER could be found.
      
Looks like CMake cannot find c/c++ compilers, since you are using windows, you would need to install MSVC. You can install it with visual studio. https://visualstudio.microsoft.com/vs/features/cplusplus/

Vaibhavmulti, 2022-10-31:
@fxia22 I already have MSVC installed on my computer.

Vaibhavmulti, 2022-11-06:
@cgokmen Any suggestions

cgokmen, 2022-11-07:
When you do `cmake --help` what do you see? Also I would try installing conda and installing cmake through conda rather than a systemwide windows cmake.

Vaibhavmulti, 2022-11-07:
Uninstalled the systemwide cmake and installed it using conda as you suggested. Still failing to build wheel for igibson.

cmake --help
[cmakehelp.txt](https://github.com/StanfordVL/iGibson/files/9950640/cmakehelp.txt)

log
[log.txt](https://github.com/StanfordVL/iGibson/files/9950644/log.txt)

cgokmen, 2022-11-07:
Can you try removing the `igibson/examples/ros` directory and doing `pip install -e .` again?

Vaibhavmulti, 2022-11-07:
After removing the ros directory 

[log.txt](https://github.com/StanfordVL/iGibson/files/9954390/log.txt)

cgokmen, 2022-11-07:
Run `pip install cmake` and try again

Vaibhavmulti, 2022-11-07:
>pip install cmake
Looking in indexes: https://pypi.org/simple, https://pypi.ngc.nvidia.com
Requirement already satisfied: cmake in c:\users\vaibh\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (3.24.1.1)

cgokmen, 2022-11-07:
Can you investigate your environment a bit (e.g. are you correctly calling the conda pip/python/cmake?) - it seems like when you run pip install it's not seeing cmake, which is not an iGibson issue and can probably be better solved by some google searches

msuv08, 2023-03-23:
@Vaibhavmulti Not sure if this is still relevant to you, but I was also running into the same issue. I'm on Ubuntu 20.04. Here's how I fixed it:
```python
conda install cmake
```
For some reason, like @cgokmen mentioned, `pip` isn't able to see your regular `cmake` installation. Thus, you need to install it through `conda` (if you're trying to clone the repository) and it should work just fine.