StanfordVL/iGibson #248: BehaviorRobot issue when use_tracked_body set to false

sujaygarlanka, 2022-11-07:
The robot misaligns the body with the hands and head when the user_tracked_body parameter is false. Also, the body falls as if it is disconnected from the hands and head. Neither happens when user_tracked_body is true. The picture attached shows how the robot is rendered in the beginning. Do you know why this may be the case or is it a bug?

I am trying to have the robot move about the space using an Oculus joystick, so I assume that setting this parameter to false is required.

[image]

cgokmen, 2022-11-07:
Hi, this indeed does look like a bug. The fact of the matter is that we have not used and/or tested the use_tracked_body = False version of this robot in quite a while. You can try to change the below two lines: 

* https://github.com/StanfordVL/iGibson/blob/3ad2aefabcf1f370ef3dabf704d0ac3becce0df9/igibson/robots/behavior_robot.py#L279 to `if True` (e.g. keep the tracked version)
* https://github.com/StanfordVL/iGibson/blob/3ad2aefabcf1f370ef3dabf704d0ac3becce0df9/igibson/robots/behavior_robot.py#L746 to `if False` (e.g. keep the tracked version)

So that even in the untracked mode, the tracked URDF and parameters are used. Hopefully this should solve your issue - normally the untracked body features slightly different offsets etc. but I don't think this will matter too much for your use case.

We can leave the issue open as a reminder for us to fix this later, but we don't have much bandwidth right this moment.

sujaygarlanka, 2022-11-08:
@cgokmen Thank you for the response. The fix works in that the body is now attached to the head and arms and moves around with them when the VR headset is attached. Also, it renders correctly before VR is attached, but once it is attached the robot renders incorrectly as above. Do you know why the rendering changes VR attaches?

cgokmen, 2022-11-08:
I'll get a fix in for this later, but for now can you try going into this line:
https://github.com/StanfordVL/iGibson/blob/3ad2aefabcf1f370ef3dabf704d0ac3becce0df9/igibson/simulator_vr.py#L473

and changing the `hmd_pos` to be `hmd_pos - np.array(behavior_robot.EYE_LOC_POSE_TRACKED[0])` instead? perhaps that will help.

cgokmen, 2022-11-08:
behavior_robot in this case is the igibson.robots.behavior_robot module.

sujaygarlanka, 2022-11-08:
Cool, that worked. Thanks!