StanfordVL/iGibson #249: Question: Can you interact with an EmptyScene with loaded objects or does the scene have to be an InteractiveIndoorScene to interact with in VR?

sujaygarlanka, 2022-11-14:


cgokmen, 2022-11-14:
You can interact with it in EmptyScene. Try the VR demo with this line replaced with an EmptyScene: https://github.com/StanfordVL/iGibson/blob/3ad2aefabcf1f370ef3dabf704d0ac3becce0df9/igibson/examples/vr/vr_simple_demo.py#L45

sujaygarlanka, 2022-11-14:
@cgokmen I tried this. Does not work because it does not have some properties that VR simulators seem to require. Getting the following error 
`Traceback (most recent call last):
  File "vr_simple_demo.py", line 111, in <module>
    main()
  File "vr_simple_demo.py", line 97, in main
    s.step()
  File "c:\users\icaro\igibson\igibson\simulator_vr.py", line 255, in step
    vr_system_dur = self.step_vr_system()
  File "c:\users\icaro\igibson\igibson\simulator_vr.py", line 213, in step_vr_system
    self.vr_system_update()
  File "c:\users\icaro\igibson\igibson\simulator_vr.py", line 324, in vr_system_update
    wall_ids = [bid for x in self.scene.objects_by_category["walls"] for bid in x.get_body_ids()]
AttributeError: 'EmptyScene' object has no attribute 'objects_by_category'`

sujaygarlanka, 2022-11-14:
I simple replaced 1 with 2
1. ` scene = InteractiveIndoorScene("Rs_int", load_object_categories=["walls"], load_room_types= ["kitchen"] )`
 2. `scene = EmptyScene()`

cgokmen, 2022-11-14:
Yeah there might be a few of these bugs that you might need to resolve. For that particular one, you can change line 324 to be something like `wall_ids = [bid for x in self.scene.objects_by_category["walls"] for bid in x.get_body_ids()] if hasattr(self.scene, "objects_by_category") else []`. I'll put this on our todo list to fix later. Or you can use an InteractiveIndoorScene (e.g. the default Rs_int) with all of its non-floor objects filtered out, e.g. `scene = InteractiveIndoorScene("Rs_int", load_object_categories=["floors"])`

cgokmen, 2022-11-14:
The latter might be less error-prone

sujaygarlanka, 2022-11-14:
I tried filtering out all elements other than floors earlier for the InteractiveIndoorScene and it doesn't seem to work. The walls and ceilings and everything still show up.

cgokmen, 2022-11-15:
I see. Yeah it's true that structural elements cannot be filtered out this way, my bad. You can feel free to try the first fix I mentioned. There might be a few more lines you might need to take out. We are having a busy month on our side so I'm unable to make any of these fixes myself but please let me know if there is anything I can do to make this easier.

sujaygarlanka, 2022-11-15:
Sorry to bring up so many issues. Thanks for all the help. The empty scene fix is working well. There is only one last issue where the BehaviorRobot is not physically translate around the scene. This issue also existed in the InteractiveIndoorScene. Not sure why this is the case.

Also, what is the difference between running iGibsonEnv as an environment and setting the mode as VR vs explicitly importing the scene and simulator as shown in the vr_simple_demo.py?

cgokmen, 2022-11-16:
There shouldn't be much of a difference honestly. The iGibsonEnv is a little bloated because it contains a lot of the RL requirements (termination condition checking, possibly different observation space, etc.) - when configured correctly it should be possible to make it behave exactly as the explicit importing does, but the default configuration probably involves some overhead.

sujaygarlanka, 2022-11-18:
Hello @cgokmen, One more issue was encountered when interacting with an object placed in an Empty Scene. When I try to grasp an object (a handle on a drawer), I get the following error:

```
Traceback (most recent call last):
  File "lsi_3d_motion_planning_main_loop.py", line 503, in <module>
    run_example(args)
  File "lsi_3d_motion_planning_main_loop.py", line 483, in run_example
    main_loop(nav_env, human, robot, bowlpans)
  File "lsi_3d_motion_planning_main_loop.py", line 366, in main_loop
    human.apply_action(actionStep)
  File "C:\Users\icaro\3d_lsi_2\igibson\robots\robot_base.py", line 446, in apply_action
    self._deploy_control(control=control, control_type=control_type)
  File "C:\Users\icaro\3d_lsi_2\igibson\robots\manipulation_robot.py", line 384, in _deploy_control
    self._handle_assisted_grasping(control, control_type)
  File "C:\Users\icaro\3d_lsi_2\igibson\robots\manipulation_robot.py", line 1058, in _handle_assisted_grasping
    self._establish_grasp(arm=arm, ag_data=self._ag_data[arm])
  File "C:\Users\icaro\3d_lsi_2\igibson\robots\manipulation_robot.py", line 952, in _establish_grasp
    and ag_bid in self.simulator.scene.objects_by_id
AttributeError: 'EmptyScene' object has no attribute 'objects_by_id'
```
Is this another bug or an issue on my end? Do I need to extend Empty Scene to include these methods to keep track of objects in it? Thanks.

cgokmen, 2022-11-19:
It is another bug - none of the VR stuff was ever tried or tested with the EmptyScene and the EmptyScene is lacking many features that it should not. You can replace `ag_bid in self.simulator.scene.objects_by_id` with `any(ag_bid in obj.get_body_ids() for obj in self.simulator.scene.get_objects())`

sujaygarlanka, 2022-11-21:
I think that code has to be replaced like below for a temporary fix because a few clauses need to be updated.

```
       objects_by_id = {}
        for obj in self.simulator.scene.get_objects():
            for id in obj.get_body_ids():
                objects_by_id[id] = obj
            
        # Create a p2p joint if it's a child link of a fixed URDF that is connected by a revolute or prismatic joint
        if (
            ag_link != -1
            and p.getJointInfo(ag_bid, ag_link)[2] in [p.JOINT_REVOLUTE, p.JOINT_PRISMATIC]
            and ag_bid in objects_by_id
            and hasattr(objects_by_id[ag_bid], "fixed_base")
            and objects_by_id[ag_bid].fixed_base
        ):
```