StanfordVL/iGibson #312: Colorized mesh's in gui_interactive

sycz00, 2023-02-03:
Hey,

I would like to know if there's any chance I can see the real textures of the objects in an igibson_scene using the gui mode ? I could not find any flag for it so far.

Thanks :)

cgokmen, 2023-02-03:
You should normally be seeing these in gui_interactive (not the pybullet GUI right?). Can you send a screenshot of what you get instead?

sycz00, 2023-02-06:
Hey,
no you are right, I use gui_interactive but everything is colored (I think it has been changed for iGibson 2.0) in the earlier versions I could see the real mesh files of the objects. Is there any way to get back to this ?
Tahnks

mjlbach, 2023-02-06:
The textures are now encrypted, the pybullet gui cannot display the encrypted textures so it shows  solid primary colors. You can send a screenshot to confirm but this is almost certainly your issue, and there is not a fix other to use only the cv2 based iG viewer (we can't add our decryption pipeline to pybullet).

sycz00, 2023-02-06:
Okay still thanks. Would have been nice for illustrations in a paper. Perhaps it can be added in the future :)

mjlbach, 2023-02-06:
The regular iG viewer should work fine for this purpose.

sycz00, 2023-02-06:
alright :)

closed 2023-02-06 by sycz00