StanfordVL/iGibson #323: python -m igibson.examples.environments.env_nonint_example

TJU-ZM, 2023-03-29:
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/thy/anaconda3/envs/igibson/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)

ChengshuLi, 2023-03-29:
Is your machine attached to a monitor? Or are you using a headless server?

ChengshuLi, 2023-04-03:
You can either set `headless=True` in the line below or run the code on a machine that has a display. 

https://github.com/StanfordVL/iGibson/blob/7d9598278b95ec1039e1f4366c66b45346521e44/igibson/examples/environments/env_nonint_example.py#L13

closed 2023-04-03 by ChengshuLi