StanfordVL/iGibson #324: Qt plugin issue

oliviaylee, 2023-04-01:
When running `python -m igibson.examples.environments.env_nonint_example` I encounter the following error (running on a cluster)

```
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/iris/u/oliviayl/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 1 is not available for rendering
qt.qpa.plugin: Could not find the Qt platform plugin "offscreen" in "/iris/u/oliviayl/anaconda3/envs/igibson/lib/python3.8/site-packages/cv2/qt/plugins"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)
```
How should I download the offscreen Qt platform plugin?

ChengshuLi, 2023-04-03:
This is because you are running on a cluster machine without a display (monitor) connected to it. 

You can either set `headless=True` in the line below or run the code on a machine that has a display. 

https://github.com/StanfordVL/iGibson/blob/7d9598278b95ec1039e1f4366c66b45346521e44/igibson/examples/environments/env_nonint_example.py#L13

closed 2023-04-03 by ChengshuLi

oliviaylee, 2023-04-03:
Hi Chengshu, thanks. If I installed iGibson via pip how should I set this? Or should I compile it from source instead?

oliviaylee, 2023-04-24:
Hi Chengshu, thanks for your help. With headless rendering, is it possible to get an output of what would have been rendered (e.g. as a gif)? For instance the image showed [here](https://stanfordvl.github.io/iGibson/quickstart.html)