StanfordVL/iGibson #327: Error to load the Qt platform plugin "xcb"

hellkonig, 2023-04-24:
To whom it may concern,

I have encountered some issues while trying to run the demo test for iGibson. I was able to successfully install iGibson and load it in ipython without any problems. However, when I tried to run the demo test using the command 
`python -m igibson.examples.environments.env_nonint_example`, I received the following error message:
```
torch is not available, falling back to rendering to memory (instead of tensor)
********************************************************************************
Description:
    Creates an iGibson environment from a config file with a turtlebot in Rs (not interactive).
    It steps the environment 100 times with random actions sampled from the action space,
    using the Gym interface, resetting it 10 times.
    ********************************************************************************
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/home/wan325/miniconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 1 is not available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Device 2 is available for rendering
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/wan325/miniconda3/envs/igibson/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)
```
I have also installed the latest version of Qt (QMake version 3.1 and Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu), so I don't think the issue is related to the Qt installation.

Additionally, I tried to test the iGibson installation by following the instructions to run `pytest --ignore disabled --ignore benchmark`, but I received the following error message:
```
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --cov=igibson --cov-report=xml
  inifile: /home/wan325/Documents/projects/active_object_detection/iGibson/pyproject.toml
  rootdir: /home/wan325/Documents/projects/active_object_detection/iGibson
```
I would appreciate any assistance in resolving these issues so that I can continue working with iGibson. Thank you for your time and attention.

ChengshuLi, 2023-04-24:
This might be because you are running on a cluster machine without a display (monitor) connected to it. 

You can either set `headless=True` in the line below or run the code on a machine that has a display. 

https://github.com/StanfordVL/iGibson/blob/7d9598278b95ec1039e1f4366c66b45346521e44/igibson/examples/environments/env_nonint_example.py#L13

hellkonig, 2023-04-26:
Hi Chengshu, thank you very much for your suggestion. I  tried setting headless=True in the line as you suggested, and it worked perfectly fine when running the code in headless mode.

However, I also tried to run the code on my laptop without headless mode and faced another error in the following:
```
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/home/wan325/miniconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 1 is not available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Device 2 is available for rendering
ERROR: Failed to EGL with glad.
```
That is why I unset the $DISPLAY variable. Could you please advise me on how to solve this issue if I want to run the code not in headless mode?

Thank you for your time and support.

linkct, 2023-07-09:
Same here. I ran the docker command `docker run --env="DISPLAY" --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" --env="QT_X11_NO_MITSHM=1" --gpus all -ti --rm igibson/igibson` on a local Ubuntu 20.04 workstation, but running `python -m igibson.examples.environments.env_nonint_example` inside the container gave me the following error:
```
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/miniconda/envs/igibson/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)
```
I can confirm that the `DISPLAY` environment variable is indeed set to `:0`, which seems to be the source of the problem, but running GUI applications outside the container is completely normal (e.g. `glxgears`), so I'm not sure how to fix this.
Thanks for any help!

unnikrishnanrnair, 2023-10-05:
running this on the host machine
`xhost +local:docker`
before starting the docker container and running the example script worked for me.