StanfordVL/iGibson #333: ERROR:  Failed to EGL with glad.

zhangcj13, 2023-07-26:
I have encountered some issues while runing python -m igibson.examples.environments.env_nonint_example, I received the following error message:
` _   _____  _  _
(_) / ____|(_)| |
 _ | |  __  _ | |__   ___   ___   _ __
| || | |_ || || '_ \ / __| / _ \ | '_ \
| || |__| || || |_) |\__ \| (_) || | | |
|_| \_____||_||_.__/ |___/ \___/ |_| |_|

********************************************************************************
Description:
    Creates an iGibson environment from a config file with a turtlebot in Rs (not interactive).
    It steps the environment 100 times with random actions sampled from the action space,
    using the Gym interface, resetting it 10 times.
    ********************************************************************************
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Device 1 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/home/cjzhang/workspace/ebi/iGibson/igibson/render/mesh_renderer/build/test_device', '2']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 2 is not available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/home/cjzhang/workspace/ebi/iGibson/igibson/render/mesh_renderer/build/test_device', '3']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 3 is not available for rendering
ERROR: Failed to EGL with glad.
`
my egl info：
‘libEGL_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
        libEGL_nvidia.so.0 (libc6) => /lib/i386-linux-gnu/libEGL_nvidia.so.0
        libEGL_mesa.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_mesa.so.0
        libEGL.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so.1
        libEGL.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so
’
please let me know what could cause this issue , how could i fix it?

Thanks!

calcualatexzy, 2023-11-05:
I have the exact same problem here. 
Have you solved? or can anyone help?

NikoHsu, 2024-02-03:
> I have the exact same problem here. Have you solved? or can anyone help?
请问你现在解决了吗

shu1ong, 2024-02-22:
I also met this problem. 
And I've tried check the egl and downgrading the nvidia driver from 525 to 470.
Make no sense. I still can not launch the demo.

kunminlinn, 2024-03-05:
same problem...

NikoHsu, 2024-03-06:
I tackled this issue by reinstall the ubuntu system, while I compare the links between the error enviroment and the executable enviroment by run `ldconfig -p | grep EGL`, and here is the difference:
error enviroment:
```
libEGL_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
**libEGL_nvidia.so.0 (libc6) => /lib/i386-linux-gnu/libEGL_nvidia.so.0**
libEGL_mesa.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_mesa.so.0
libEGL.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so.1
libEGL.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so
```

executable enviroment: 
```
libEGL_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
libEGL_mesa.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_mesa.so.0
libEGL.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so.1
libEGL.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so
```

by run `ldconfig -p | grep libnvidia-glcore`
error enviroment:
```
libnvidia-glcore.so.550.40.07 (libc6,x86-64) => /lib/x86_64-linux-gnu/libnvidia-glcore.so.550.40.07
libnvidia-glcore.so.550.40.07 (libc6) => /lib/i386-linux-gnu/libnvidia-glcore.so.550.40.07

```
executable enviroment: 
`libnvidia-glcore.so.545.23.06 (libc6,x86-64) => /lib/x86_64-linux-gnu/libnvidia-glcore.so.545.23.06
`

I hope this helps！