StanfordVL/iGibson #344: Unable to pass tests with official docker containers(igibson/igibson:lates，igibson/igibson-vnc:v2.0.6)

chenyinlin1, 2023-11-01:
To whom it may concern.
I installed the officially released docker container on my own Linux server. And tested it according to the iGibson documentation([(https://stanfordvl.github.io/iGibson/quickstart.html)]). But except for simple commands like import igibson, there are other commands involving the GUI.
Currently the container can be accessed and the gpu detected as follows：
[image]
But when I run the command: python -m igibson.examples.environments.env_nonint_example,  the following error is reported:
[image]
Please ask how to solve this problem. Thank you very much.

chenyinlin1, 2023-11-02:
Using the command sudo glxinfo | grep "OpenGL" , I found out that there was a problem with the nvidia driver on the server, where the driver for opengl was not downloaded.  
[image]
Re-download the driver files and install them, don't bring "-no-opengl-files" in the command, so that only the driver files will be installed, not the OpenGL files. 
[image]
Rerun the command "sudo glxinfo | grep "OpenGL"" and found that opengl supports 4.6

chenyinlin1, 2023-11-02:
Although the opengl version issue was resolved, the iGibson GUI still cannot be displayed.
Follow the igibson instructions documentation to open the docker container:
docker run --env="DISPLAY" --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" --env="QT_X11_NO_MITSHM=1" --gpus all -ti --rm igibson/igibson:latest
Run the command and the following error is reported:
[image]

ChengshuLi, 2023-11-06:
Is it a headless server (i.e. the server isn't connected to a display monitor)? If so, this behavior is expected. You can set `mode="headless"`.