StanfordVL/iGibson #353: Building wheel for igibson (pyproject.toml) ... error

tangliangcjy, 2024-02-02:
I am getting these erros while installing igibson. I have also attached the text file for entire log.Please let me know how to proceed.
note: This error originates from a subprocess, and is likely not a problem with pip.
ERROR: Failed building wheel for igibson

ERROR: Could not build wheels for igibson, which is required to install pyproject.toml-based projects

cgokmen, 2024-02-02:
Please paste the whole output of the `pip install` call. This does not show the actual error.

tangliangcjy, 2024-02-03:
> Please paste the whole output of the `pip install` call. This does not show the actual error.
installing in window，the error：
 error: [Errno 2] No such file or directory: 'D:\\gitcode\\robots\\iGibson\\igibson\\render\\openvr\\bin\\win64\\openvr_api.dll'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson, which is required to install pyproject.toml-based projects

tangliangcjy, 2024-02-03:
> Please paste the whole output of the `pip install` call. This does not show the actual error.

installing in window :
 error: [Errno 2] No such file or directory: 'D:\\gitcode\\robots\\iGibson\\igibson\\render\\openvr\\bin\\win64\\openvr_api.dll'
      [end of output]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson, which is required to install pyproject.toml-based projects

tangliangcjy, 2024-02-03:
> Please paste the whole output of the `pip install` call. This does not show the actual error.

Installing in ubuntu20.04:
        File "/tmp/pip-build-env-8trztm0r/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "<string>", line 54, in run
        File "<string>", line 107, in build_extension
        File "/home/d5/anaconda3/envs/igibson/lib/python3.8/subprocess.py", line 364, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j2']' returned non-zero exit status 2.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson, which is required to install pyproject.toml-based projects

tangliangcjy, 2024-02-03:
[ 86%] Building CXX object CMakeFiles/EGLRendererContext.dir/cpp/mesh_renderer.cpp.o
      In file included from /tmp/pip-install-qpc3ztkh/igibson_5f959cc847e248bc916ea832d2fc3cde/igibson/render/cpp/mesh_renderer.h:12:0,
                       from /tmp/pip-install-qpc3ztkh/igibson_5f959cc847e248bc916ea832d2fc3cde/igibson/render/cpp/mesh_renderer.cpp:36:
      /usr/local/cuda-10.1/include/cuda_gl_interop.h:66:10: fatal error: GL/gl.h: No such file or directory
       #include <GL/gl.h>
                ^~~~~~~~~
      compilation terminated.
      CMakeFiles/EGLRendererContext.dir/build.make:103: recipe for target 'CMakeFiles/EGLRendererContext.dir/cpp/mesh_renderer.cpp.o' failed
      make[2]: *** [CMakeFiles/EGLRendererContext.dir/cpp/mesh_renderer.cpp.o] Error 1
      make[2]: *** Waiting for unfinished jobs....
      CMakeFiles/Makefile2:133: recipe for target 'CMakeFiles/EGLRendererContext.dir/all' failed
      make[1]: *** [CMakeFiles/EGLRendererContext.dir/all] Error 2
      make[1]: *** Waiting for un

tangliangcjy, 2024-02-03:
Traceback (most recent call last):
        File "/home/d5/anaconda3/envs/igibson/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/d5/anaconda3/envs/igibson/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/d5/anaconda3/envs/igibson/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "/tmp/pip-build-env-8trztm0r/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 404, in build_wheel
          return self._build_with_temp_dir(
        File "/tmp/pip-build-env-8trztm0r/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 389, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-bu

zby1218, 2024-03-21:
I also encountered this error, have you solved it？

tangliangcjy, 2024-04-06:
> I also encountered this error, have you solved it？

yes ， do you need any help now？

ck567-wqm, 2024-04-10:
> > I also encountered this error, have you solved it？
> 
> yes ， do you need any help now？

@tangliangcjy I also encountered this error,Can you tell me the solution? Thank you

zby1218, 2024-04-11:
> > I also encountered this error, have you solved it？
> 
> yes ， do you need any help now？

yes, could you tell me how you solve this problem?

tangliangcjy, 2024-04-11:
> > > I also encountered this error, have you solved it？
> > 
> > 
> > yes ， do you need any help now？
> 
> @tangliangcjy I also encountered this error,Can you tell me the solution? Thank you

what is your system platform?

tangliangcjy, 2024-04-11:
> > > I also encountered this error, have you solved it？
> > 
> > 
> > yes ， do you need any help now？
> 
> yes, could you tell me how you solve this problem?

what is your system platform?

zby1218, 2024-04-11:
> > > > I also encountered this error, have you solved it？
> > > 
> > > 
> > > yes ， do you need any help now？
> > 
> > 
> > yes, could you tell me how you solve this problem?
> 
> what is your system platform?

my system platform is windows