StanfordVL/iGibson #355: ig_assets version 2.2.2 incompatible. Needs to be in range [2.0.6, 2.2.1)

taylorlulu, 2024-02-25:
When I'm trying to run the igibson 1.4.0, it failed due the following errer:
[image]

```
ig_assets version 2.2.2 incompatible. Needs to be in range [2.0.6, 2.2.1)
```
I can't find where to download the older vision of ig_assets. How can I download the older version of ig_assets?

nblkStudy, 2024-04-25:
Could you solve it? I'm facing the same issue.

taylorlulu, 2024-04-25:
@nblkStudy I solved the problem by directly modify the file, "ig_assets version 2.2.2 incompatible. Needs to be in range [2.0.6, 2.2.1)" to "ig_assets version 2.2.2 incompatible. Needs to be in range [2.0.6, 2.2.2)". It worked out then. I'm not sure will it cause some further issue.

cgokmen, 2024-04-26:
It seems to me as though the current version on our `main` branch (e.g. `2.2.2`) is compatible with the `2.2.2` version of the assets, no? Why do you need to use the 2.2.0 version of iGibson?

taylorlulu, 2024-04-27:
@cgokmen Maybe the problem comes from https://github.com/StanfordVL/iGibson/blob/7d9598278b95ec1039e1f4366c66b45346521e44/igibson/__init__.py#L101-L114. As I'm using the older version of iGibson. So the MAX_ASSETS_VERSION_EXCL may be smaller than 2.2.2. However the only version available is 2.2.2.

cgokmen, 2024-04-27:
I think removing this check for those versions particularly won't break anything, but why do you need to be at an older version of iGibson? Both you and @nblkStudy .

taylorlulu, 2024-04-27:
@cgokmen I'm using iGibson 2.2.0. Since the request of repository I'm working on is to install iGibson==2.2.0. I'm not sure if I'm using the latest of version of iGibson will bring some potential Issues.