StanfordVL/iGibson #360: 3D FRONT to igibson

tractormaster4596, 2024-05-16:
I try to use the file process_scene_3dfront.sh to transport 3D front scene data to the type of igibson. During the process, i occur some troubles：
First is:
Traceback (most recent call last):
  File "step_4_convert_to_ig.py", line 265, in <module>
    main()
  File "step_4_convert_to_ig.py", line 261, in main
    convert_scene(scene_name, args.source, select_best=args.select_best)
  File "step_4_convert_to_ig.py", line 88, in convert_scene
    cat_dir = get_ig_category_path(obj_category)
  File "D:\work\iGibson\igibson\utils\assets_utils.py", line 151, in get_ig_category_path
    assert category_name in os.listdir(ig_categories_path), "Category {} does not exist".format(category_name)
AssertionError: Category chair does not exist

Seond is :
I try to transport the 3D objects which the scene needs,like chair and table. Then i run step_4_convert_to_ig.py success,but step_5 run,the erro info is:
Traceback (most recent call last):
  File "step_5_generate_trav_map.py", line 78, in <module>
    main()
  File "step_5_generate_trav_map.py", line 73, in main
    generate_trav_map(scene_name, args.source, False)
  File "step_5_generate_trav_map.py", line 35, in generate_trav_map
    scene._set_first_n_objects(3)
  File "D:\work\iGibson\igibson\scenes\igibson_indoor_scene.py", line 715, in _set_first_n_objects
    raise ValueError(
ValueError: The _set_first_n_object function is now deprecated due to incompatibility with recent object state features. Please use the load_object_categories method for limiting the objects to be loaded from the scene.

The environment of my pc is win10, python=3.8, without cuda and the 3D front json file is 0a8d471a-2587-458a-9214-586e003e9cf9.json.

And I have some questions about it.
That is what is the correct processes of transport 3D Front scene data to igibson type?  Whether did i transport the 3D objects beforehand which the scene needs.

Thanks for apply!

ZhuSiiqii, 2025-01-17:
Hi,i have the same question, may i ask you whether you have fixed it and how? 
Many thanks!

1ssb, 2025-07-27:
Has a solution to this been developed?